badgerloop-software/pod-embedded #26: Jenkins Automated Testing

ezraboley, 2019-02-08:
Very much a stretch goal, however our code quality would benefit if we were to get an automated testing environment setup. The general idea is to have some test scripts that just run periodically to make sure all of our code still builds, and to make sure that we still pass all of our tests.

vkottler, 2019-02-08:
You want [Travis CI](https://travis-ci.org/) and a good first job to run is some kind of code format check / linter (i.e. for C/C++, something like [clang-format](https://clang.llvm.org/docs/ClangFormat.html).

vkottler, 2019-02-08:
@EUdds really useful thing to have for linting JavaScript as well if anyone has time to play around with it

ezraboley, 2019-02-08:
Oh nice, I wasn't really sure between the ones that I found. I'll look into that instead. Definitely going to throw clang on there, and I might test out writing some scripts in pytest that it could run

rdaruwala, 2019-03-06:
Implemented in PR #59 , closing

closed 2019-03-06 by rdaruwala