badgerloop-software/pod-embedded #76: DS3231 RTC Module Driver

ezraboley, 2019-04-23:
We are getting a real time clock (RTC) so that offline building doesn't raise errors and to let us timestamp our packets to keep track of latency.

This will require a driver to set and read the time along with a simple API to do so from the rest of our existing code base. [Attached here is the datasheet.](https://datasheets.maximintegrated.com/en/ds/DS3231.pdf)

There is also a potential stretch goal of giving the clock Watchdog capabilities (it has an alarm interrupt output), however this should only be undertaken after the main task has been completed.

rdaruwala, 2019-04-23:
https://learn.adafruit.com/adding-a-real-time-clock-to-beaglebone-black/overview

Should be a simple process. We can just set the system time and the programs will be able to pull it automatically

rdaruwala, 2019-06-13:
@ezraboley Is this complete?

ezraboley, 2019-06-13:
It is not. And I was reminded of how useful it would be when I kept running into clock skew issues at Steamfitters. Planning on taking care of it soon