badgerloop-software/pod-embedded #80: Retro Sensors Drivers + Voting

rdaruwala, 2019-05-12:
Add retro drivers as well as voting logic to ensure that the `retroCount` value in the data structure is accurate

Data values to update:
- `time_t lastRetro1, lastRetro2, lastRetro3`
- `int retroCount`

rdaruwala, 2019-05-12:
We'll also need to implement IMU checking + timer logic as described in the safety package

ezraboley, 2019-05-12:
Im pretty far into this one on the `embedded-retro-dev` branch. There is a beaglebone GPIO API there along with code that captures the retro asynchronously. The only thing left is some wonky timestamp issues I was running into with the nanosecond counter, and the IMU checking

ezraboley, 2019-05-30:
Voting is ready for testing, will test before merging in (and closing) the retro-dev branch

ezraboley, 2019-06-13:
The voting ~seems~ to be working, I'd like to do a full nav test though so that can be moved to a new issue

closed 2019-06-13 by ezraboley